barbeau/gpstest #574: Status works only with "Interval=1 sec", if you config "Interval>1 sec", #Sats drops to ZERO and no Sats List

bernieEV, 2021-12-16:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Status' Screen
2. Tap on 'config'
3. Scroll down to 'Interval' and set a value above 1 sec
4. See 'Status' Screen

**Expected behavior**
I expect battery savings but still get a list of Sats

**Observed behavior**
the #Sats value drops down to ZERO and no Sats are listed

my environment: EUROPE and I have selected GALILEO only.
the OS is Android12 on a fresh google PIXEL6
the app is from google play_store "GPStest" v3.9.16(18093-google)

barbeau, 2021-12-16:
@bernieEV Thanks for the feedback! If you show all satellites (i.e. remove Galileo filter) do you see any in the list (for example, GPS)? Also, do you have "Force full GNSS measurements" enabled in Settings? And what exact interval value are you testing with?

bernieEV, 2021-12-17:
Hello Sean,

I tested with values 10 sec, 15 sec,
but more helpful seems to be the change from 1 sec to 2 sec.

It is Android 12 and "Force full GNSS measurements" is enabled.
no filter , show all satellites as requested

default Interval 1 sec shows in     #Sats 16/25/25
changing Interval to 2 sec show in #Sats  8/13/13

doubling the time interval changes the satellites used to half
some magic math happens here

best regards and good hunt
Bernie

barbeau, 2021-12-17:
@bernieEV Thanks for the info! So to summarize, it seems like Galileo satellites disappear when you increase the interval above 1 second? Do any other satellites seem to disappear or is it just Gallileo?

If you could provide before and after screenshots showing all satellites (no filter) at 2 second interval that would be helpful. Also screenshots of your system settings for About that's shows the exact software build version of Android you're using. I could then report it to Google as a bug. That seems strange, I don't know why the device would do that.

bernieEV, 2021-12-17:
Here are the screenshots
Screenshot_20211217-141745.png   details about Android12
GPStest Status Screen with sorted Constellation, signal strength
Screenshot_20211217-141943.png   1 sec interval , all is visible
Screenshot_20211217-143149.png   2 sec interval , EUROPE and CHINA disappeared

[image]
[image]
[image]
d

barbeau, 2021-12-18:
Reported to Google at:
https://issuetracker.google.com/issues/211144840

Please star the issue if you'd like Google to fix it.

barbeau, 2021-12-28:
@bernieEV note that Google has requested additional info if you're able to provide it directly on the issue:
https://issuetracker.google.com/issues/211144840#comment3

botanegg, 2022-03-03:
Fixed in  google issue tracker https://issuetracker.google.com/issues/211144840#comment8

barbeau, 2022-03-03:
Thanks for the head sup @botanegg!

closed 2022-03-03 by barbeau