barbeau/gpstest #576: accuracy screen diagrams auto scaling labels wrong

bernieEV, 2021-12-27:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Accuracy Screen'
2. Pull up 'Diagrams'
3. See 'lower Diagram, vertical deviation'
4. the auto-scaling is "lazy", see list: Zero line, 0m, 1m, 2m, 2m, 2m, 3m

**Expected behavior**
A clean linear scale

**Observed behavior**
see appended screenshot

**GPStest v3.9.16 on Android 12 (stock version) on Google PIXEL 6 ** 

What version of GPSTest did you encounter this on (see "Help->About")? What make and model device (e.g., Samsung Galaxy S10) did you encounter this on?  What Android version (e.g., Android 10.0) are you running?  Is it the stock version from the manufacturer or a custom ROM?  What version of Google Play Services is running on your device?

**Screenshots:** 

Can be created by pressing the Volume Down and Power Button at the same time on Android 4.0 and higher.
[image]

bernieEV, 2021-12-28:
if the full scale is less then 6 m, we need one digit after the decimal point.
the unit "meters" could be placed after the label.

barbeau, 2022-01-05:
@bernieEV Thanks for flagging this. It's a good problem to have, because it only happens at very low error values 😄. I'm using a library for the graph so I'm not sure how much control I have over how this renders with small values, but I'll take a look.