barbeau/gpstest #578: order of colomns in STATUS view

bernieEV, 2022-01-12:
**Is your feature request related to a problem? Please describe.**
my way of thinking and reading tables ...

**Describe the solution you'd like**
different order of colomns in STATUS view.
the real global space is the frequency spectrum and modulation  (CF - colomn 1)
next is signal strength at receiver (C/N0 - colomn 2)
next is nation (GNSS national flags - colomn 3)
next is index inside nation (ID - colomn 4)
next is (Flags - colomn 5)
next is horizontal plane (Azim - colomn 6) 
next is vertical (Elev - colomn 7)

**Describe alternatives you've considered**

**Additional context**
just my thoughts as an enhancement request
the ID is definitely a subunit of GNSS, there are duplicates in different constellations.

barbeau, 2022-01-12:
Thanks for the suggestion! I'll consider this, although given the history of the app that goes way back users are also used to the existing order.

bernieEV, 2022-08-02:
Hi Sean,
I got a timeout email.
I still prefer to order and sort objects from "big to little"

closed 2023-09-07 by stale[bot]