barbeau/gpstest #580: Bit of a Noob

Bandit253, 2022-01-18:
Hi, I am new to Android development but have the need to create a GPS logger type app and was thinking of using this excellent project as a basis of at least learning how to interface.

I have downloaded but I can't seem to get it to run in Android Studio. 

My usecase is a location referencing (field collection app) for assets, it is just a POC to assess if my proposed methodology would work.
None of the other 'examples' don't have the details this app already has.  Truth be known I'd prefer to do it in python, but the interface is not mature enough.

barbeau, 2022-01-19:
@Bandit253 welcome to Android!

>I have downloaded but I can't seem to get it to run in Android Studio.

Can you be more specific about the problem you're seeing? Screenshots and logs would help. What are you seeing and what do you expect to see? What version of Android Studio are you using?

You can check out https://github.com/barbeau/gpstest/blob/master/BUILD.MD#building-in-android-studio as well, although the prerequisites are dated at this point - you shouldn't need to install the Android SDK separately, you can just install Android Studio.

Bandit253, 2022-01-19:
Thanks for the warm welcome.
I had another 'fresh' go and I have managed to get it going.  I was just about to close this issue anyway.
Thanks so much for the speedy response.

Bandit253, 2022-01-19:
cheers

closed 2022-01-19 by Bandit253