barbeau/gpstest #581: Add satellite names

barbeau, 2022-01-25:
**Is your feature request related to a problem? Please describe.**
A user request:

>Please add information about the specific name a particular satellite, for example,  Galileo-IOV FM3.

[image]

https://github.com/barbeau/gpstest/issues/220 has some links for name sources.

PeterCullenBurbery, 2022-09-10:
I would like to express my support for adding GPS names.

I would like to find the Shriya satellite that joined Galileo on 29 August 2022 with GPS Test.

barbeau, 2022-09-10:
There is a start to cataloging some of the names in this method:
https://github.com/barbeau/gpstest/blob/master/GPSTest/src/main/java/com/android/gpstest/util/SatelliteUtils.java#L41

barbeau, 2023-10-29:
Not stale

PeterCullenBurbery, 2023-10-29:
Thank you for marking the issue as not stale.

PeterCullenBurbery, 2024-05-22:
This is valid.