barbeau/gpstest #586: Can't upload device capabilities

kroeckx, 2022-04-20:
On a Samsung A22, when trying to share the device capabilities, I get the message: `Please switch to the Status screen, wait for a GNSS fix, and then return to this screen to upload device capabilities.".

The device has a fix. Several of the satellites are marked with an "U" flag, but no other flags.

barbeau, 2022-04-20:
@kroeckx Thanks for reporting this! It appears to be the same issue at https://github.com/barbeau/gpstest/issues/495. I believe I've fixed this in the code as part of the overhaul of the app I've been working on, but I haven't done a new release that contains this fix (a few other items need to be completed first).

I'm going to close this issue because the problem's been resolved in the code. If you'd like to be among the first to test the fix when I do a release, please join the beta testing group at https://play.google.com/apps/testing/com.android.gpstest!

kroeckx, 2022-04-20:
I've installed the version from the play store now and joined the beta program. Let me know if I should test it.

barbeau, 2022-04-20:
You'll have a bit of waiting time, I still need to finish some things before I do the next release.