barbeau/gpstest #593: Log uses scientific notation instead of decimal notation.

tarquinwj, 2022-07-18:
**Describe the bug**
https://github.com/barbeau/gpstest/issues/553 introduced a bug that only comes into effect in specific locations. If your latitude, longitude, altitude, accuracy, vertical accuracy, speed, etc, are very close to 0 but not quite 0, they will get logged in scientific notation, which confuses software that processes logs.

Instead of 0.00000000000000001, the logs will contain 1E-17

Testing this requires setting up a mocked location with those specific coordinates and accuracies. I tested using "Location Changer - Fake GPS", since it was the only one I could find that let me specify exact coordinates.

**To Reproduce**
1. Use an app or debugging environment that allows you to specify the mocked location to whatever precision you want.
2. Set the mocked location to 0.00000000000000000001 latitude, longitude, altitude.
3. If possible, also set the other values that are logged, such as the accuracy, vertical accuracy, speed accuracy, speed, bearing accuracy, bearing, etc.
4. Enable logging in GPSTest
5. Check the contents of the log

**Expected behavior**
Log should contain 0.00000000000000000001 or just 0.000000000000000 (15 decimal places)

**Observed behavior**
Log will contain 1E-20

**App, Device and Android version:** 

3.9.16 (18093-google)
Android 12

tarquinwj, 2022-07-18:
Possibly the mock location app is feeding this number format into the mocked location, I don't know.

FWIW, I have fixed this sort of issue in one of my own projects using this:

	$basicString = $num.''; //cast to string
	if( strpos( $basicString, 'E' ) === false ) {
		//simple number, didn't need scientific notation
		return $basicString;
	} elseif( strpos( $basicString, 'E-' ) ) {
		//1.2345E-15
		preg_match( "/\.(\d+)E-(\d+)$/", $basicString, $matches );
		return sprintf( '%.' . ( strlen( $matches[1] ) + $matches[2] * 1 ) . 'F', $num );
	}
	//1.2345E+15
	return sprintf( '%.0F', $num );

So it uses sprintf, just like the old GPSTest code used to, but it works out if it needs to first, and if it does, then it sets the specific number of decimal places to match the number of decimal places that are needed to include the entire natural precision of the number.

tarquinwj, 2022-07-18:
# Location fix format:
#   Fix,Provider,LatitudeDegrees,LongitudeDegrees,AltitudeMeters,SpeedMps,AccuracyMeters,BearingDegrees,UnixTimeMillis,SpeedAccuracyMps,BearingAccuracyDegrees,elapsedRealtimeNanos,VerticalAccuracyMeters
...
Fix,gps,1.0E-20,1.0E-20,1.0E-15,2.7777777,1.0,0.0,1658162530754,,,2594706345002284,

I couldn't work out how to spoof the other variables, so I am hoping you can test those somehow in your development environment. (Location mocking apps all really suck, very badly!)

tarquinwj, 2022-07-18:
Managed to get a few more. Speed and accuracy too:
Fix,gps,1.0E-20,1.0E-20,1.0E-15,2.7777778E-16,1.0E-15,0.0,1658164076254,,,2596251845677945

barbeau, 2022-07-18:
Thanks @tarquinwj! Just knowing the values you're using and the fields should be enough for me to reproduce in the unit tests and fix the issue, so this is very helpful.

barbeau, 2022-07-18:
Looks like the root issue is that Kotlin string templates:

```kotlin
"Fix,$provider,$latitude,$longitude,$altitude,$speed,$accuracy,$bearing,$time,"
```

...convert certain floats and doubles to scientific notation using `toString()` by default. The fix seems to be to use `toBigDecimal().toPlainString()` for each value:
https://stackoverflow.com/questions/57755536/kotlin-how-to-convert-double-to-string-in-kotlin-without-scientific-notation

closed 2022-07-18 by barbeau