barbeau/gpstest #598: SBAS ID mappings need to be updated

barbeau, 2022-08-17:
**Describe the bug**
See https://www.gps.gov/technical/prn-codes/L1-CA-PRN-code-assignments-2021-Jun.pdf.

For example, PRN 120 is no longer assigned to EGNOS

**To Reproduce**
Steps to reproduce the behavior:
See https://github.com/barbeau/gpstest/blob/master/GPSTest/src/main/java/com/android/gpstest/util/SatelliteUtil.kt#L284.

**Expected behavior**
ID mapping should be update to date

**Observed behavior**
ID mapping is old

yakMM, 2022-08-19:
@barbeau Outside of updating the ID mapping, should the missing types be added with the corresponding flags, strings and entry in the FAQ?

Are missing today types for:
- ASECNA – Agency for Aerial Navigation Safety in Africa and Madagascar (PRN 120 & 147)
- SPAN – Southern Positioning Augmentation Network (AUS-NZ) (PRN 122)
- ASAL – Algerian Space Agency (PRN 148)
- QZSS – Quazi-Zenith Satellite System (several PRNs > 182)

barbeau, 2022-08-23:
@yakMM That would be great. Early on I was waiting for SBAS to actually show up in devices because device support wasn't clear, but I think we're at the point now where we can go ahead and add the officially documented satellites.

barbeau, 2022-08-23:
See also the links to code that need to be updated in this thread - https://github.com/barbeau/gpstest/pull/599#issuecomment-1224012834

davecollett, 2022-08-23:
New PR here: https://github.com/barbeau/gpstest/pull/601

danieldjewell, 2023-02-19:
@barbeau For what it's worth, GPSTest does show that it's receiving signal from SBAS/WAAS (PRNs 131, 133)... Although it only shows a signal (C/N0) - no flags. 

I don't know if the Android APIs really give a lot of info on whether or not SBAS is actually being *used*...

barbeau, 2023-02-19:
>Although it only shows a signal (C/N0) - no flags.

Thanks - yes, this is the expected behavior for "unsupported" signals in GPSTest. The record will still show up in the list if the device supports it, but it just won't have a flag.

>I don't know if the Android APIs really give a lot of info on whether or not SBAS is actually being used...

Yes, this is something Android isn't really clear on - see https://github.com/barbeau/gpstest/issues/267#issuecomment-473483888.

barbeau, 2023-04-04:
I'm changing this ticket to focus on adding the new SBAS constellations that aren't currently supported, and https://github.com/barbeau/gpstest/issues/618 will track updating the PRN assignments for the SBAS we currently support.

barbeau, 2023-04-20:
SouthPAN is now implemented (#601 - thanks @davecollett!).

Still TODO:
* ASECNA – Agency for Aerial Navigation Safety in Africa and Madagascar (PRN 120 & 147)
* SPAN – Southern Positioning Augmentation Network (AUS-NZ) (PRN 122)
* ASAL – Algerian Space Agency (PRN 148)
* KASS - Korean Augmentation Satellite System (PRN 134) - https://www.gps.gov/technical/prn-codes/L1-CA-PRN-code-assignments-2021-Jun.pdf, https://www.airbus.com/en/newsroom/press-releases/2022-06-airbus-built-measat-3d-communications-satellite-ready-for-launch

davecollett, 2023-04-20:
Just for clarity - SouthPAN is the service previously referred to as SPAN in the PRN Code assignments under PRN 122

barbeau, 2023-04-20:
Ah, right, thanks @davecollett. I should read more carefully before I copy and paste 😁.

Narugakuruga, 2023-09-10:
Looks like BDS is testing their SBAS service.
(PNR 130, 143, 144) The band is B1C and B2a.
https://www.csno-tarc.cn/system/constellation

barbeau, 2023-09-19:
Here is a new SBAS PRN update as of April 2023:
https://www.gps.gov/technical/prn-codes/

barbeau, 2023-10-24:
Example of KASS in the wild:
https://groups.google.com/d/msgid/gpstest_android/138e8b2b-dd59-4941-85ab-5e4cf7035379n%40googlegroups.com

gfanini, 2025-06-24:
excuse me if off topic 

Do you know any list of gps chipset in android cellphones,how to find out which ones would have Broadcom GNSS for Galileo EGNOS position augmentation compensation broadcast, if not RTK