barbeau/gpstest #600: Take a picture and add GPS data to the exif metadata of the image option for app

PeterCullenBurbery, 2022-08-23:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
My phone does not add all the GPS data available to the metadata of a picture for later analysis. I'm able to extract some of the GPS metadata, but some data is missing even though the phone has access to it like GPS satellites in view and GPS speed.
**Describe the solution you'd like**
I would like to request a feature where you can take a picture with the app and it will add all the GPS data extracted with the application to the metadata of the picture stored in the EXIF metadata file. I would like to request the following GPS metadata elements:
"GPSAltitude", "GPSAltitudeRef", "GPSAreaInformation", \
"GPSDateStamp", "GPSDestBearing", "GPSDestBearingRef", \
"GPSDestDistance", "GPSDestDistanceRef", "GPSDestLatitude", \
"GPSDestLatitudeRef", "GPSDestLongitude", "GPSDestLongitudeRef", \
"GPSDifferential", "GPSDOP", "GPSImgDirection", "GPSImgDirectionRef", \
"GPSLatitude", "GPSLatitudeRef", "GPSLongitude", "GPSLongitudeRef", \
"GPSMapDatum", "GPSMeasureMode", "GPSPositioningError", \
"GPSProcessingMethod", "GPSSatellites", "GPSSpeed", "GPSSpeedRef", \
"GPSStatus", "GPSTag", "GPSTimeStamp", "GPSTrack", "GPSTrackRef", \
"GPSVersionID"

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
There is an app iNaturalist that I think displays the gps precision when you take a picture of a plant.

barbeau, 2022-08-23:
Here is a good API doc that lists all the Exif tags that are supported:
https://developer.android.com/reference/androidx/exifinterface/media/ExifInterface

PeterCullenBurbery, 2022-08-23:
I looked at the webpage. I don't know how to program an Android app.

barbeau, 2022-08-23:
@PeterCullenBurbery no worries 😁. It's mainly a note to whoever takes us on in the future.

PeterCullenBurbery, 2022-08-29:
I have an example with a lot of GPS calculations at https://www.wolframcloud.com/obj/04327028-71ea-4d5b-a3d1-f4be1e522195

PeterCullenBurbery, 2022-08-29:
I will note that not all of the gps document is related to gps files, some is related to Mathematica, but I have some helpful information present.

closed 2023-10-03 by stale[bot]