barbeau/gpstest #603: no map in accuracy view

FadeFx, 2022-09-09:
**Describe the bug**
when i open accuracy view there is only my location point and a rastwr, but no map visible, when i open map view it is visible.

[image]

guidot, 2024-10-21:
Can confirm. The accuracy circle and the © text in the lower left is displayed.

Map view works normally.

OS: Android 14, Custom ROM, MicroG
Device: Realme XT

barbeau, 2024-10-22:
@FadeFx @guidot thanks for the report! Do you have the Fdroid version or the build from Google Play?

FadeFx, 2024-10-22:
As you can see, my report is from 2022, I just tested and can confirm that it works now...
Fdroid version...

guidot, 2024-10-22:
Mine is from f-droid too. Version is the current 3.10.3.

guidot, 2024-10-22:
Can you please re-open?

barbeau, 2024-10-22:
Looks like this is a duplicate of https://github.com/barbeau/gpstest/issues/316. If you're not seeing map tiles, it's likely that we've exhausted our free tile quota of the service we're using.