barbeau/gpstest #607: Don't show user result of injecting/clearing assist data

barbeau, 2022-10-24:
**Describe the bug**

In the past GPSTest has used the returned boolean value of `LocationManager.sendExtraCommand()` to show a Toast to the user on the success or failure of injecting time and PSDS/XTRA data as well as clearing assist data:
https://developer.android.com/reference/android/location/LocationManager#sendExtraCommand(java.lang.String,%20java.lang.String,%20android.os.Bundle)

However, the latest version of the docs say for the returned value:

>true always, the return value may be ignored

This value has always been suspect, and now we have confirmation that we shouldn't rely on it. So we should no longer show the user success or failure messages based on this returned value.

**To Reproduce**
Steps to reproduce the behavior:
1. Inject XTRA/PSDS or time data or clear assist data using option in navigation drawer

**Expected behavior**

Do not show the user success or failure message - just acknowledge that GPSTest has attempted the action.

**Observed behavior**
We show the user a success or failure message as a Toast 

**App, Device and Android version:** 

N/A

zztzztzzt, 2022-10-26:
in Find X5 pro, if I directly download the apk and run, ths clearing assist data (TTFFs 30+s) can work, but if I download the source code and build apk, the clearing assist data doesnot work(TTFF 2+s). I don't know what happen

barbeau, 2022-10-28:
@zztzztzzt Hmmm, strange - I wouldn't expect a release or debug build to have any difference in results. Maybe it just happened to be the circumstances when you tried it? Could you try it again?

If the Find X5 pro is a Snapdragon chipset, it might not work. I recently tested another Snapdragon device in clearing the assist data didn't work, there was an immediate fix.

barbeau, 2022-11-03:
The problem here is that the app doesn't actually know if the framework executed the command. It can submit the command, but the framework could ignore it.

zztzztzzt, 2022-11-09:
I think it's a frame problem, I uninstalled and reinstalled the app once from F-Droid, the clear assist data button didn't work (TTFF 2+s), but when I clicked on clear data a dozen times continuously, it worked, and every time after that it worked（TTFF 20+s）

commit 271708ab (2023-04-12): fix: Don't indicate if time/data injection/deletion request succeeded

Fixes #607
closed 2023-04-12 by barbeau