barbeau/gpstest #608: Share --> Google maps requires a different style

ValZapod, 2022-11-23:
Marker of coordinates is not displayed in google maps, while it does display in other apps from Yandex or 2GIS. It is displayed when you click on google maps https://geohack.toolforge.org/geohack.php?params=50.66212424;27.50909286

Style of link is https://www.google.com/maps?ll=50.662124,27.509093&q=50.662124,27.509093&hl=en&t=m&z=11

**To Reproduce**
Steps to reproduce the behavior:
1. Go to share
2. Tap on arror
3. Press on Maps (of Google)
4. See no red marker

**Expected behavior**

No bug in google maps.

**Observed behavior**
A clear and concise description of what actually happened. Since you work on Google maps you can fix that Android API standard link inside Gapps, or add a new icon in share menu.

**App, Device and Android version:** 

Android 13 gakaxy S22 Ultra 0 model snapdragon

v3.9.16 app.

**Screenshots:** 

What I want:
[image]

ValZapod, 2023-01-09:
@barbeau Did you see?

barbeau, 2023-01-12:
@ValZapod Thanks for the feature request for Google Maps!  I'm working on the Android location team currently, so unfortunately the Maps app is a completely different group. 

Could you please open a feature request at https://issuetracker.google.com/issues?

That's the best place for this request to get attention. And if you can post the link to the issue after you've created it here I'll see if I can find the right person for it. Because this isn't specific to GPSTest I'm going to close this issue.

closed 2023-01-12 by barbeau

ValZapod, 2023-01-12:
So there is no way to change the style of link for google maps? Cause this does work: https://www.google.com/maps?ll=50.662124,27.509093&q=50.662124,27.509093&hl=en&t=m&z=11

barbeau, 2023-01-12:
Not to my knowledge