barbeau/gpstest #614: GALILEO HAS started 2023-01-24

bernieEV, 2023-01-28:
Hi Sean,
I am in Europe/Austria and got GALILEO navigation info in the news.

please check this service started 2023-01-24
https://www.gsc-europa.eu/galileo/services/galileo-high-accuracy-service-has

Is it possible to add this to GPStest app ?
thanks

bernieEV, 2023-01-29:
Hi Sean,
I checked my HW.
It is "Google PIXEL6" and your program "GPStest" tells in the HELP "Broadcom BCM4776"
looking into the Broadcom specs it can receive european GALILEO frequency E1 and E5a.
looking into the GALILEO frequency plan the new service HAS is in the frequency E6b.
unfortunately my PIXEL6 cannot receive E6b.
best regards
Bernie

danieldjewell, 2023-02-19:
While @barbeau is definitely more of an expert than I am, my own experience with Android devices is that there is not *necessarily* a correlation between what a receiver chip might support (on paper) and what it will support, as configured by an OEM.

For example, my US-model Samsung S22 Ultra has a Snapdragon 8 Gen 1 ... which, according to the datasheet, supports "GPS, Glonass, BeiDou, Galileo, QZSS, NavIC" and "Dual Frequency GNSS (L1/L5)" ... When I run GPSTest, I see zero BeiDou, and only L1 Glonass. (Admittedly, it does say "L1/L5" and L2 Glonass is on a very different frequency from GPS/Galileo L1/L5...)

Without digging deep into the hardware configuration, some of the features might be disabled/inaccessible - even though the _chip_ might be capable...

barbeau, 2023-02-19:
Sorry for the delayed response - @danieldjewell is right, there are a lot of reasons why the specific hardware specs might differ from the ultimate phone features. I actually wrote an article about this:
https://barbeau.medium.com/crowdsourcing-gnss-capabilities-of-android-devices-d4228645cf25

@bernieEV unfortunately yes, Galileo HAS is on E6b, which is a frequency phones don't support.

closed 2023-02-19 by barbeau