barbeau/gpstest #615: Dark mode toast messages are shown with white text on a white background

altayhunter, 2023-01-30:
**Describe the bug**
The toast messages (popup pills near the bottom) in dark mode are shown with white text on a white background, making them almost unreadable.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable "Use Dark Theme" in settings.
2. Tap on "Inject PSDS data" or "Inject time data".
3. See toast with white text on a white background.

**Expected behavior**
Popup message should have white text on a black background.

**Observed behavior**
Popup message has white text on a white background, making it practically unreadable.

**App, Device and Android version:** 

GPSTest v3.9.16 (18093-google)  
Observed on a OnePlus 7 Pro running Pixel Experience Plus 13, and also on a Google Pixel 5 running stock Android  
Android version 13

**Screenshots:** 

[image]

JonnyTech, 2023-01-30:
It may be an issue with your system, this is from my LineageOS 20 / Android 13:

[image]

barbeau, 2023-01-31:
@altayhunter thanks for the report! Are you using the app installed from Google Play or building the app yourself from the main branch?

altayhunter, 2023-01-31:
I'm using the app installed from Google Play.

barbeau, 2023-01-31:
Thanks, I can reproduce this as well using the version downloaded from Google Play on a Pixel 7 with Android 13. It looks fine though on a Samsung Galaxy S21 Plus with Android 13, so not sure what's going on here yet.

paulle, 2023-04-07:
I can confirm this bug.
Android 13
Sony experia 5 III
Gpstest downloaded from f-droid