barbeau/gpstest #617: [Feature request] adding NTRIP support to access Galileo High Accuracy Service PPP over Internet

ohault, 2023-03-12:
Galileo High Accuracy Service (HAS) is now available and also via Internet.

Galileo HAS increases the accuracy of Galileo to sub-meter levels.
https://www.gsc-europa.eu/galileo/services/galileo-high-accuracy-service-has/internet-data-distribution

davecollett, 2023-04-20:
My understanding is that these corrections would have to be applied at a lower level than the app - they'd need to be applied as part of the positioning algorithm on the chipset..

danieldjewell, 2023-05-27:
My understanding is that the Galileo HAS service is providing PPP-type correction data (whether delivered over the E6 channel via satellite or the internet) and not RTK correction data - RTK is much more accurate, but is also limited to a much smaller eographic area (the accuracy decreases with distance). Typically, with an RTK setup, you have a stationary "base" that creates/calculates the correction data that then feeds that data to a "rover" where the GNSS chip receives the corrections data (via bluetooth, wifi, LoRa, the internet, etc.) and incorporates that into its position solution  

That said, regardless of the type of correction data, it *is* possible to perform the corrections in software and not directly on the GNSS chip. There's even an Android app for it (see: https://github.com/jancelin/RTKlibDroid) and RTKlib in general ... (If *real time* accuracy is not required, corrections can be applied in post-processing, too.) I don't recall what level of "source" data is required to do this - I'm not sure that a log in NMEA format is enough.....  

However, for the highest accuracy (and performance/speed), the correction data is typically fed into the GNSS chip itself. (See the various -and excellent- u-Blox high-accuracy GNSS receivers ...) 

Lastly: Because of the relatively high level of abstraction between the GNSS chips in Android devices (the hardware itself) and the OS/Userland, there are some -challenges- regarding that.

barbeau, 2023-06-15:
Yes, currently GPSTest doesn't compute the GNSS position itself - it just shows the result of the GNSS position computed by the chipset.

However, I'd like to eventually implement a true positioning engine within GPSTest and support things like HAS, so I'll leave this issue open.

barbeau, 2024-01-02:
Not stale

gdt, 2024-04-12:
My take is that software RTK is vastly out of scope for 'gpstest' and ought to be separate, so this feature doesn't belong.

By separate, I can see another app doing RTK, and perhaps gpstest plumbing raw data to, and getting nmea from.   But it seems not really reasonable to add rtk code to 'use the api and show what it is doing'.

issteve, 2025-03-25:
There is now a free GPS enhancement PPP-RTK to a cm-precision from the German government (and the states) which comes in software for Android (and Windows ...) - 
SAPOS / GEPOS. 

More about it (in German) at https://gepos.sapos.de/nutzung

gdt, 2025-03-25:
> There is now a free GPS enhancement PPP-RTK to a cm-precision from the German government (and the states) which comes in software for Android (and Windows ...) - SAPOS / GEPOS.
> 
> More about it (in German) at https://gepos.sapos.de/nutzung

Is the software open source, and can you share a link to the sources?

issteve, 2025-03-26:
> Is the software open source, and can you share a link to the sources?

It's not directly mentioned there. If barbeau is interested in this I'm willing to investigate more about it.