barbeau/gpstest #620: Map screen lacks required OSM attribution (F-Droid build)

gdt, 2023-04-07:
**Describe the bug**
In the F-Droid build, the map tab shows OSM data and lacks the required attribution.

**To Reproduce**
Steps to reproduce the behavior:
1. Install from F-Droid main repo
2. In hamburger menu, select Map

**Expected behavior**
The OSM map should have attribution per 
https://www.openstreetmap.org/copyright

**Observed behavior**
No attribution, and a fortiori no clickable link to the license.

**App, Device and Android version:** 

GPSTest v3.9.16
CalyxOS based on AOSP 13

** Comment **

It's likely quite clear from source code that attribution isn't attempted, so I'm not going to go into further details proving it is missing unless that' s actually in dispute.

gdt, 2023-04-07:
@AntoninDelFabbro your comment about attribution is valid in substance, even if coming across as non-collaborative in tone.  I myself am an osm editor, and often cranky about the license, but I reserve serious crankiness for people attempting to get something for themselves, vs people contributing to the free software ecosystem.

Perhaps you know how to fix, and could prepare a PR?

barbeau, 2023-04-07:
@gdt @AntoninDelFabbro Thanks for flagging this, it fell off my radar and got mixed in with the issue of supporting local OSM tiles. I'll add the license attribution in the next release.

commit d91808b0 (2023-04-12): fix: Add OSM attribution notices to F-Droid flavor maps

Fixes #620