barbeau/gpstest #627: GPSTest Won't Open on Moto E4 Plus - Android 7.1.1

vkbellis, 2023-04-14:
**Describe the bug**
GPSTest v3.10.0 won't open on Moto E4 Plus (XT1770) - Android 7.1.1

**To Reproduce**
Unable to open app on this old device

**Expected behavior**
GPSTest would open like a normal app.

**Observed behavior**
Initial white screen with blue top information bar background, displays for about 2 seconds, followed by a very brief flash, GPSTest has stopped.

**App, Device and Android version:** 
GPSTest v3.10.0, Moto E4 Plus (XT1770) - Android 7.1.1

barbeau, 2023-04-20:
@vkbellis Thanks for the report! Could you please capture a bug report after this crash happens (https://developer.android.com/studio/debug/bug-report#bugreportdevice) and share it?

vkbellis, 2023-04-20:
Hi Sean, 
I just tried running another test (unsuccessfully) on the Moto E4 Plus, followed by the requested generation of a bug report, then this repeating notice:

Google Play services isn't responding
Close app
Wait
Send Feedback

Tapping: Wait, Wait, Wait, eventually was able to send an email from the Android device to myself... which still still hasn't arrived... until just now! 

Hope this helps. - Kind regards, Kelly

[bugreport-NMA26.42-162-1-2023-04-20-16-25-58.zip](https://github.com/barbeau/gpstest/files/11290298/bugreport-NMA26.42-162-1-2023-04-20-16-25-58.zip)

barbeau, 2023-04-20:
Thanks!! From the bug report, seems like this may be the relevant error:

```
04-14 11:23:13.186 10530 10530 E AndroidRuntime: FATAL EXCEPTION: main
04-14 11:23:13.186 10530 10530 E AndroidRuntime: Process: com.android.gpstest, PID: 10530
04-14 11:23:13.186 10530 10530 E AndroidRuntime: java.lang.AbstractMethodError: abstract method "void android.location.LocationListener.onProviderDisabled(java.lang.String)"
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at android.location.LocationManager$ListenerTransport._handleMessage(LocationManager.java:310)
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at android.location.LocationManager$ListenerTransport.-wrap0(LocationManager.java)
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at android.location.LocationManager$ListenerTransport$2.handleMessage(LocationManager.java:249)
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:110)
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:203)
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6339)
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1084)
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:945)
```

barbeau, 2023-04-20:
Well, there is also this one:

```
04-20 16:25:39.070 13399 13420 E AndroidRuntime: Process: com.android.gpstest, PID: 13399
04-20 16:25:39.070 13399 13420 E AndroidRuntime: java.lang.NullPointerException: sensorManager.getDefault‚Ä¶(Sensor.TYPE_ORIENTATION) must not be null
04-20 16:25:39.070 13399 13420 E AndroidRuntime: 	at com.android.gpstest.library.data.SharedSensorManager$_sensorUpdates$1.invokeSuspend(SharedSensorManager.kt:139)
04-20 16:25:39.070 13399 13420 E AndroidRuntime: 	at com.android.gpstest.library.data.SharedSensorManager$_sensorUpdates$1.invoke(SharedSensorManager.kt)
04-20 16:25:39.070 13399 13420 E AndroidRuntime: 	at com.android.gpstest.library.data.SharedSensorManager$_sensorUpdates$1.invoke(SharedSensorManager.kt)
04-20 16:25:39.070 13399 13420 E AndroidRuntime: 	at kotlinx.coroutines.flow.ChannelFlowBuilder.collectTo$suspendImpl(Builders.kt:322)
04-20 16:25:39.070 13399 13420 E AndroidRuntime: 	at kotlinx.coroutines.flow.ChannelFlowBuilder.collectTo(Builders.kt)
04-20 16:25:39.070 13399 13420 E AndroidRuntime: 	at kotlinx.coroutines.flow.CallbackFlowBuilder.collectTo(Builders.kt:336)
04-20 16:25:39.070 13399 13420 E AndroidRuntime: 	at kotlinx.coroutines.flow.internal.ChannelFlow$collectToFun$1.invokeSuspend(ChannelFlow.kt:60)
04-20 16:25:39.070 13399 13420 E AndroidRuntime: 	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
04-20 16:25:39.070 13399 13420 E AndroidRuntime: 	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
04-20 16:25:39.070 13399 13420 E AndroidRuntime: 	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:570)
04-20 16:25:39.070 13399 13420 E AndroidRuntime: 	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:749)
04-20 16:25:39.070 13399 13420 E AndroidRuntime: 	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:677)
04-20 16:25:39.070 13399 13420 E AndroidRuntime: 	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:664)
04-20 16:25:39.070 13399 13420 E AndroidRuntime: 	Suppressed: kotlinx.coroutines.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@4a32304, Dispatchers.Default]
```

barbeau, 2023-04-20:
Looks like this is the fix for the 1st stack trace - https://stackoverflow.com/a/67900719/937715.

vkbellis, 2023-04-26:
Issue still seems to be present in GPSTest v3.10.1 - see attached bug report.

[bugreport-NMA26.42-162-1-2023-04-26-10-47-03.zip](https://github.com/barbeau/gpstest/files/11334339/bugreport-NMA26.42-162-1-2023-04-26-10-47-03.zip)

barbeau, 2023-04-27:
@vkbellis Thanks for the bug report, I'll take a look

barbeau, 2023-04-27:
That's strange, I'm still seeing the same issue with the method `LocationListener.onProviderDisabled()` even though it's implemented in v3.10.1. I wonder if it has something to do with using Kotlin...

```
04-14 11:23:13.186 10530 10530 E AndroidRuntime: FATAL EXCEPTION: main
04-14 11:23:13.186 10530 10530 E AndroidRuntime: Process: com.android.gpstest, PID: 10530
04-14 11:23:13.186 10530 10530 E AndroidRuntime: java.lang.AbstractMethodError: abstract method "void android.location.LocationListener.onProviderDisabled(java.lang.String)"
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at android.location.LocationManager$ListenerTransport._handleMessage(LocationManager.java:310)
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at android.location.LocationManager$ListenerTransport.-wrap0(LocationManager.java)
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at android.location.LocationManager$ListenerTransport$2.handleMessage(LocationManager.java:249)
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:110)
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:203)
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6339)
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1084)
04-14 11:23:13.186 10530 10530 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:945)
```

barbeau, 2023-04-27:
We could try using LocationListenerCompat - https://b.corp.google.com/issues/190695723.

closed 2023-04-27 by barbeau

barbeau, 2023-04-28:
@vkbellis I'm rolling out v3.10.2 now that I'm hoping will fix this. Could you please give it a try?

barbeau, 2023-04-28:
@vkbellis Could you also please try to run the latest GnssLogger release (https://play.google.com/store/apps/details?id=com.google.android.apps.location.gps.gnsslogger) and let me know if you see the same crash there?

commit 3cea066d (2023-04-27): fix: Use LocationListenerCompat to avoid crashes on API <=30

Fixes #627