barbeau/gpstest #628: Increase Measure accuracy's displayed precision

vkbellis, 2023-04-14:
**Is your feature request related to a problem? Please describe.**
Not so much of a problem beyond stemming the question of the stored value that GPSTest uses.

**Describe the solution you'd like**
Ideally, when the user enters a value to a given precision for the Measure Accuracy baseline position, GPSTest should reciprocate that precision in its display; e.g., the ellipsoid height was entered as 21.522 m, but the *saved* value that's displayed by GPSTest gets rounded/ or truncated to 21.5 m. Similarly, though to a smaller degree, lat and lon decimal.degree values entered to 8 places to the right of the decimal were trimmed down to 7.

This truncated/ rounded display *appears* to only be an issue when returning to the station. The entered value remained displayed just as entered during the initial accuracy test.

barbeau, 2023-04-24:
@vkbellis Thanks for flagging this! This was an oversight on my part, I didn't intend the cap the precision of entered values when the app is closed. I'm fixing this so lat/lon get saved to 10 decimal places, and alt gets saved to 4.

closed 2023-04-24 by barbeau

vkbellis, 2023-04-24:
@barbeau Thanks for fixing  this Sean! :)

commit f7567f1a (2023-04-24): fix: Don't cap precision of entered benchmark coordinates

Fixes #628