barbeau/gpstest #630: F-Droid can't build

licaon-kter, 2023-04-19:
Wear libs are not FOSS

ref: https://github.com/barbeau/gpstest/commit/7f56a0e8490a0de77c9cce5b67936461aea2b079

licaon-kter, 2023-04-19:
related: https://github.com/barbeau/gpstest/pull/631

Fixed via: https://gitlab.com/fdroid/fdroiddata/-/commit/308018f389babbf215e66b31c33a650eb0026384

Works mostly ok, gets a lock and etc, but seen one error too (when the app was put in background)
```
java.lang.RuntimeException: Unable to pause activity {com.android.gpstest.osmdroid/com.android.gpstest.ui.MainActivity}: java.lang.NullPointerException
	at android.app.ActivityThread.performPauseActivityIfNeeded(ActivityThread.java:4744)
	at android.app.ActivityThread.performPauseActivity(ActivityThread.java:4695)
	at android.app.ActivityThread.handlePauseActivity(ActivityThread.java:4646)
	at android.app.servertransaction.PauseActivityItem.execute(PauseActivityItem.java:46)
	at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:176)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:97)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2067)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:223)
	at android.app.ActivityThread.main(ActivityThread.java:7705)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:952)
Caused by: java.lang.NullPointerException
	at com.android.gpstest.ui.sky.SkyFragment.getBinding(SkyFragment.kt:79)
	at com.android.gpstest.ui.sky.SkyFragment.onGnssStopped(SkyFragment.kt:238)
	at com.android.gpstest.ui.sky.SkyFragment.access$onGnssStopped(SkyFragment.kt:72)
	at com.android.gpstest.ui.sky.SkyFragment$trackingListener$1.invoke(SkyFragment.kt:105)
	at com.android.gpstest.ui.sky.SkyFragment$trackingListener$1.invoke(SkyFragment.kt:105)
	at com.android.gpstest.library.util.PreferenceUtil.newStopTrackingListener$lambda$0(PreferenceUtil.kt:218)
	at com.android.gpstest.library.util.PreferenceUtil.$r8$lambda$_QFM2gjZjNGM7s1BOw8LuZsLtqA(Unknown Source:0)
	at com.android.gpstest.library.util.PreferenceUtil$$ExternalSyntheticLambda1.onSharedPreferenceChanged(Unknown Source:4)
	at android.app.SharedPreferencesImpl$EditorImpl.notifyListeners(SharedPreferencesImpl.java:637)
	at android.app.SharedPreferencesImpl$EditorImpl.apply(SharedPreferencesImpl.java:511)
	at com.android.gpstest.library.util.PreferenceUtils.saveBoolean(PreferenceUtils.java:104)
	at com.android.gpstest.library.util.PreferenceUtils.saveTrackingStarted(PreferenceUtils.java:235)
	at com.android.gpstest.ForegroundOnlyLocationService.unsubscribeToLocationUpdates(ForegroundOnlyLocationService.kt:253)
	at com.android.gpstest.ui.MainActivity.onPause(MainActivity.kt:394)
	at android.app.Activity.performPause(Activity.java:8168)
	at android.app.Instrumentation.callActivityOnPause(Instrumentation.java:1511)
	at android.app.ActivityThread.performPauseActivityIfNeeded(ActivityThread.java:4734)
	... 12 more
```

closed 2023-04-19 by licaon-kter

barbeau, 2023-04-19:
@licaon-kter Thanks for flagging this and fixing it for F-Droid!