barbeau/gpstest #633: Copy log files to Download directory when logging is complete

barbeau, 2023-04-20:
**Is your feature request related to a problem? Please describe.**
As discussed in https://groups.google.com/g/gpstest_android/c/FAx6M0Zj9Ug, it's not clear to some users where the file is located following the updates for scope storage in v3.10.0.

**Describe the solution you'd like**
We should copy log files to the Download directory like GnssLogger when logging is complete.

vkbellis, 2023-04-20:
Why couldn't GPSTest files be in their own discrete folder?

barbeau, 2023-04-20:
Android limited the ability of apps to create folders with the introduction of scoped storage. IIRC this also affects creating subfolders within Downloads, but I'll double-check.

vkbellis, 2023-04-20:
IICORC - If I could only recall correctly (LOL)! It's easier for me to just test stuff. For example in today's test using Geo++ RINEX Logger v2.1.6, I was able to locate my files without issue and without physically cabling phone to PC. Why does this work for Geo++ but not GPSTest?

[image]

[image]

[image]

barbeau, 2023-04-20:
>Why does this work for Geo++ but not GPSTest?

Looks like the last time that Geo++ was updated is Feb 2020:
https://play.google.com/store/apps/details?id=de.geopp.rinexlogger

The next time they update, they'll have to bring their app into compliance with Google's latest requirements. Here are the details:
https://support.google.com/googleplay/android-developer/answer/11926878

It's the same with GPSTest - v3.9.16 works fine and it can write files anywhere, but now that I'm updating the app on Google Play I need to follow the new requirements too. 

These requirements are actually what's driving the timeline for the release of v3.10.0 - if I don't update GPSTest by May 1st Google will start delisting the app for newer Android versions so those user's won't be able to download it. Same with Geo++ - if they don't update the app, then it won't be available for newer Android versions.

barbeau, 2023-04-21:
So I'm happy to report I was wrong - you can create subfolders under "Downloads", so I'm going to put files in "Downloads/GPSTest".

closed 2023-04-21 by barbeau

vkbellis, 2023-04-21:
Cool! Great to hear! Thanks for scoping this out ;)

Thank you also for the explanation.