barbeau/gpstest #638: 3.10.0 doesn't work at all. Always crash

unixasket, 2023-04-24:
**Describe the bug**
After Update via Fdroid to version 3.10.0 GPSTest doesn't start on my Ulefone Power Armor X11 pro (Android 12). If I want to start it, it crashes (always, reproducible)

**To Reproduce**
Steps to reproduce the behavior:
1.  Start GPSTest
2. App crashes

**Expected behavior**
GPSTest works again

**Observed behavior**
Crashes at every start of the App

**App, Device and Android version:** 

GPSTest Version: 3.10.0 from Fdroid
Phone Modell: Ulefone Power Armor X11 pro
Android: 12

**Screenshots:** 

not possible (app doesn't start at all)

JonnyTech, 2023-04-24:
Did you grant GPSTest the location permission?

unixasket, 2023-04-24:
Yes, of course
René Maroufi
***@***.***
Threema: 3K3YCPKT
Telegram: @ReneMarou
Matrix: @ReneM:matrix.org
XMPP: ***@***.***

barbeau, 2023-04-24:
@unixasket Thanks for reporting this! Sorry, this was a simple mistake on my part (I accidentally included a dependency that only exists on Google devices) and should be fixed in https://github.com/barbeau/gpstest/commit/2a64a40fd92047ad635d55861a2df429fc319010.

I'll get another release out today, which will be v3.10.1. Please give that a try and report back if you still have problems.

barbeau, 2023-04-27:
@unixasket Have you had a chance to try the v3.10.1 release to see if this is resolved?

unixasket, 2023-04-29:
Yes, thanks, it works.
René Maroufi
***@***.***
Threema: 3K3YCPKT
Telegram: @ReneMarou
Matrix: @ReneM:matrix.org
XMPP: ***@***.***