barbeau/gpstest #639: GPSTest 3.10 keeps on crashing

PCigales, 2023-04-28:
First of all, I would like to express my gratitude to you for having implemented the background service allowing to keep the gps fix running continuously (very useful when trekking).

The new versions unfortunately crash very often on my Galaxy A8 under Oreo, mainly during the first seconds (if the app manages to run beyond, it becomes less likely to stop working).

Here is what I have found in the bug report that may indicates the cause of the issue (that could be related to  #627):

04-28 07:36:42.419 10919 31165 31165 E AndroidRuntime: FATAL EXCEPTION: main
04-28 07:36:42.419 10919 31165 31165 E AndroidRuntime: Process: com.android.gpstest, PID: 31165
04-28 07:36:42.419 10919 31165 31165 E AndroidRuntime: java.lang.AbstractMethodError: abstract method "void android.location.LocationListener.onStatusChanged(java.lang.String, int, android.os.Bundle)"
04-28 07:36:42.419 10919 31165 31165 E AndroidRuntime: 	at com.android.gpstest.library.data.SharedLocationManager$_locationUpdates$1$callback$1.onStatusChanged(SharedLocationManager.kt:74)
04-28 07:36:42.419 10919 31165 31165 E AndroidRuntime: 	at android.location.LocationManager$ListenerTransport._handleMessage(LocationManager.java:298)
04-28 07:36:42.419 10919 31165 31165 E AndroidRuntime: 	at android.location.LocationManager$ListenerTransport.-wrap0(Unknown Source:0)
04-28 07:36:42.419 10919 31165 31165 E AndroidRuntime: 	at android.location.LocationManager$ListenerTransport$2.handleMessage(LocationManager.java:243)
04-28 07:36:42.419 10919 31165 31165 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:105)
04-28 07:36:42.419 10919 31165 31165 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:164)
04-28 07:36:42.419 10919 31165 31165 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:6944)
04-28 07:36:42.419 10919 31165 31165 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
04-28 07:36:42.419 10919 31165 31165 E AndroidRuntime: 	at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:327)
04-28 07:36:42.419 10919 31165 31165 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1374)

I hope it will be helpful to you, if I can bring more assistance, I will be happy to try, but I have very little knowledge of Android and java.

Thanks in advance.

PCigales, 2023-04-28:
I take the opportunity to ask you if your projects for enhancements include adding a feature like a HTTP REST service allowing another device to retrieve gps data from gpstest. I mean I have been working on a python/javascript app to visualize and edit gps tracks under Windows (https://github.com/PCigales/GPXTweaker) and I wondered if it would be possible to get the user location by sending a request to the phone on an address with a port set in gpstest, such as http://ip:port/gpstest?location where "ip" would be the ip or the dns solvable name of the phone and "port" choosen by the user in gpstest options, or maybe http://ip:port/gpstest returning a json or xml answer with all available real time data. There does not seem to be an established protocol for this kind of information, such as ntp for time.
I hope what i write is understandable, my english is not as good as yours...

barbeau, 2023-04-28:
@PCigales Thanks for the report, and the stack trace! I believe this is the same issue at https://github.com/barbeau/gpstest/issues/627, so I'm closing this as a duplicate. I just released v3.10.2 that's rolling out now with a change that will hopefully fix this. Could you please give v3.10.2 a try and let me know if it fixes it?

>I take the opportunity to ask you if your projects for enhancements include adding a feature like a HTTP REST service allowing another device to retrieve gps data from gpstest. 

This sound like https://github.com/barbeau/gpstest/issues/482 - please upvote that issue if you'd like to see it implemented, and feel free to comment too!

closed 2023-04-28 by barbeau

barbeau, 2023-04-28:
@PCigales Could you also please try to run the latest GnssLogger release (https://play.google.com/store/apps/details?id=com.google.android.apps.location.gps.gnsslogger) and let me know if you see the same crash there?

PCigales, 2023-04-28:
Thank you for your answer and for the fix. Despite all the the tortures that I inflicted on it, the new release did not crash, so that was the right fix.

Indeed, my suggestion is similar to #482, except that rather than uni/multi-casting datagrams, the idea is to run a TCP server to provide the data in response to requests.

PCigales, 2023-04-28:
> @PCigales Could you also please try to run the latest GnssLogger release and let me know if you see the same crash there?

The app does not crash but two tabs remain empty: "plots" and "wls analysis" despite right configuration. The logs contain data.
[image]

barbeau, 2023-04-28:
Ok, thanks for checking!

Komprogue, 2023-04-29:
It also crashes on Android 8.1, but only if app is launched with auto-start GNSS enabled and device location isn't turned on.