barbeau/gpstest #644: Show data from GnssCapabilities.getGnssSignalTypes()

rapperskull, 2023-05-17:
I'm developing a method to add L5 functionalities to my device (it is sold in different markets with different capabilities), and I would like to have a way to easily test what is available on my device. I can check if I get L5 reception, but I can't test for NavIC for example, since I'm in Europe.

I saw that there's a getGnssSignalTypes() method in GnssCapabilities, but I didn't understand what's about. If it describes what constellations and bands the device supports, it would be nice to have that information available in GPSTest.

barbeau, 2023-05-18:
@rapperskull Agreed! This new API is available on Android U and up, and I plan to add support for it to power the Dashboard view of supported signals (see work-in-progress at PR https://github.com/barbeau/gpstest/pull/570). I'm going to close this issue as a duplicate of https://github.com/barbeau/gpstest/issues/313.

Also, please keep me posted on your method to add L5 to your device!

closed 2023-05-18 by barbeau

rapperskull, 2023-05-18:
Didn't notice the "Added in Android UpsideDownCake".
By the way, this is my method to add L5 support to the Realme GT 2 Pro:
https://forum.xda-developers.com/t/mod-enable-gps-l5-e5a-b2a-bands-on-global-eu-model-rmx3301.4587471/
You can find more info on the GitHub page: https://github.com/rapperskull/realme_gps_l5_enabler