barbeau/gpstest #645: [meta] Stale bot considered harmful

gdt, 2023-05-31:
tl;dr: The stale bot is fundamentally a wrong thing.  It should be de-configured.

<rant>The stale bot is dysfunctional.  Just because an issue has not had activity does not meet it is invalid or unworthy of the space it takes up in the tracker.  I say this as someone who, as maintainer of an unrelated projects, aggressively closes non-useful issues.   if someone who cares doesn't react fast enough, a bug can be incorrectly closed.   On yet another unrelated project, I have seen issues closed, and even with commenting, not re-opened.

It's fine if the maintainer decides that an issue is so speculative that it is not worth the cognitive load.   I have on occasion scanned the entire issue list, asked for updates, and closed some -- especially those on which I have previously asked questions with no response.   But I see zero value in timers.
<rant>

barbeau, 2023-06-15:
@gdt Thanks for the feedback! I hear you, stale bot can certainly be annoying. However, whether or not it's harmful I think depends on it's use with the project, which depends on the project maintainers. I personally use it to help gauge community interest in specific issues, especially those that are typically lower priority or require bandwidth that I don't have. For example, I know which issues you're continually interested in based on you keeping them alive :). It's also a reminder to me about specific issues, which I will occasionally update with new information that I obtained since the issue was opened (and mark it not stale).

Also, I personally don't consider a closed issue to be permanently closed - they can always be re-opened. If it's closed by stalebot, that means that it's gone 400 days without comment (200 for warning and another 200 then close). To me that's a reasonable gauge that there isn't anyone interested in that feature anymore. And if someone misses the window to comment, again, we will just re-open.

That all being said, I'd certainly welcome additional feedback from others on stalebot.

gdt, 2023-06-15:
My perception is that I can't reopen an issue; only maintainers can.   I guess the big problem is that there are incomplete bug reports that nobody cares about and ones that we all agree are valid and it's just going to take ~forever to deal with them.  But it's your repo, your effort and your call.

gdt, 2025-01-24:
With time passing, there's another problem with the stale bot: Legitimate issues that are open long-term accumulate noise from close/pointless-comment/open.

Perhaps there is a way to tell it not to mark things again, that have been marked/unmarked once?

I see your point about reminders.    As a maintainer I just try to scan the tracker periodically and ping people If I think the issue shouldn't really be open.   I use the feedback label for when I've asked and they are in timeout mode (no automated, just records for me).  For issues asking from something sensible that I am just not getting to but would with 20x spare time, I leave them.  If they are conceptually legit but never going to happen even with time, I move them to a wiki page of "speculative feature requests" that lists them with links to the closed issues.

(In my case, I've tried to be respectful of your intent for the app and not open things that are  out of scope.  Actually I probably see the scope as narrower than you do...)

gdt, 2026-05-30:
It now seems that stalebot is no longer maintained and the repo has been archived.