barbeau/gpstest #646: Feature request: "screenshot mode"

LuccoJ, 2023-06-05:
**Is your feature request related to a problem? Please describe.**
Time and again, I want to show screenshots of GPSTest to other people. This is usually for comparing the GNSS performance of multiple devices, mine or theirs, show what constellations are supported, the time to first fix, the number of satellites seen and used, SBAS presence, and so on.

However, if I do it in public places, I have to work to remove my coordinates. I really cannot afford everyone to know exactly where I leave, almost to the room level. I usually leave the altitude and the alt/az of the satellites untouched, but I think in theory, someone could calculate my location from the latter, although I wouldn't be able to use the maths involved.

**Describe the solution you'd like**
There could be a "screenshot mode" that one activates before taking a screenshot, which makes GPSTest blank out the coordinates, and (optionally?) the satellites' alt/az and the altitude. I would definitely leave the number of satellites, the error, the flags, and the time to first fix.

**Describe alternatives you've considered**
I suppose it would also be possible to react to the user taking a screenshot (some apps blank themselves out when this is done), in which case I would not do this by default, but still only if an option "Blank sensitive data on screenshots" is selected in the settings. After all, not everyone takes screenshots to make them public, so the default should probably just be to take a normal screenshot.

barbeau, 2023-06-06:
@LuccoJ This is a good idea, I have the same issue frequently too. The new Dashboard screen I've been working on in PR https://github.com/barbeau/gpstest/pull/570 might address some of this as it shows a summary of device properties, but wouldn't cover everything above.

LuccoJ, 2023-06-06:
Oh, I forgot about a detail I wanted to mention: though not immediately related to hiding private information, having the device name and OS version visible when "screenshot mode" is on (or when taking a screenshot, anyway) would also facilitate the scenarios I'm thinking about.

barbeau, 2023-06-06:
Toggling VisualTransformation for the specific text fields may work as an implementation for Status:
https://developer.android.com/jetpack/compose/text#formatting

barbeau, 2023-12-23:
Not stale

closed 2025-01-26 by stale[bot]