barbeau/gpstest #647: Implement support for Galileo High Accuracy Service

gle42, 2023-06-13:
**Galileo High Accuracy Service (HAS)** provides correction data achieving an horizontal **accuracy under 20 cm (8 inches)**. It now available free of charge. This data is distributed through the E6 band (only available to high-end receivers) and through the Internet.

This request is to implement support for this Internet based High Accuracy Service:
- retrieving this data using the _Galileo HAS Internet Data Distribution_ based on the Ntrip protocol
- implementing the PPP algorithm using this data

Note that access to this data is possible to registered users only. A decision will have to be taken to decide whether each end user of the application should register in person or if some registered GPSTest backend server should get the data and redistribute it to end users (I think the license permits this).

See [details on Galieo's website](https://www.gsc-europa.eu/galileo/services/galileo-high-accuracy-service-has)
You can get detailed spec after registering.

barbeau, 2023-06-15:
Thanks @gle42! there is already an issue open for this at https://github.com/barbeau/gpstest/issues/617 so I'm going to close this as a duplicate. Feel free to comment more there.

closed 2023-06-15 by barbeau

gle42, 2023-06-16:
Thanks for replying so quickly.
I apologize for having wasted your time by creating a duplicate. I should have searched more before creating this.

barbeau, 2023-06-16:
No worries!