barbeau/gpstest #648: DeadSystemException when enabling logging to a file of all types of GNSS data

nAglTI, 2023-06-15:
**Summary**
DeadSystemException when enabling logging to a file of all types of GNSS data, after which the phone goes into reboot.

**Steps to reproduce**
Steps to reproduce the behavior:
1. Go to 'Settings'
2. Go to 'Logging and Ouput'
3. In "File Output" check the checkboxes: NMEA, Measurements, GnssStatus, Orientation, Location, Navigation Messages.
4. Go to 'Status' and wait a couple of minutes.

**Observed behavior:**
When performing the above steps for a couple of minutes, the phone may first freeze and after a while it completely freezes and reboots (see the video from the phone screen).
This happens only when all the checkboxes described above are selected, but if you select each checkbox individually, no error occurs.
The error can only be reproduced on Pixel 6a, a check was also made for Realme 6 Pro (Android 11) and Xiaomi 12T Pro (Android 13), there were no errors on them.

**App, Device and Android version:** 
GPSTest v3.10.2
Device: Google Pixel 6a (Build number - TQ2A.230505.002)
Android version: 13
Google Play Services version: 23.18.19

**Videos:** 
Video from Android Studio logcat: https://youtu.be/WRWh90Jpnfs
Pixel 6a screen video: https://youtu.be/a4HxD-MgkUo

barbeau, 2023-06-15:
Wow, that shouldn't happen! 

@nAglTI Thanks for the very specific steps to repro and the video links! Sounds like a Pixel 6a-specific issue. Could you please open an issue at https://issuetracker.google.com/issues/new?component=313183&template=0 and share the same info? And please post the link to the bug here after you do. Thanks!

nAglTI, 2023-06-15:
https://issuetracker.google.com/issues/287505109

barbeau, 2023-06-15:
Thanks @nAglTI!

barbeau, 2023-08-11:
So it looks like there was a bug in GPSTest that was spamming the inject assist data methods when logging was turned on, and this triggered the freeze and reboot. I've fixed this in https://github.com/barbeau/gpstest/commit/fd260766084ac72908de20658202f32db774add5 so assist data is only injected once per logging session, which should resolve this issue in GPSTest.

closed 2023-08-11 by barbeau

barbeau, 2023-08-15:
This is fixed in the latest v3.10.3 release that I pushed out today to Google Play, which should be available to download shortly - https://github.com/barbeau/gpstest/releases/tag/v3.10.3.

commit fd260766 (2023-08-11): fix: Only inject assist data once per logging session

Fixes #648