barbeau/gpstest #650: [Feature Request] (optionaly) disable background service on low battery / when energy saver kicks in

ChristianTacke, 2023-07-05:
**Is your feature request related to a problem? Please describe.**
I have enabled the "Run GNSS in background" option and sometimes forget to terminate the app (and the background measurements). This drains the battery somewhat and finally my battery is quite low.

**Describe the solution you'd like**
When my battery reaches 30% (my setting), the device automatically enters energy saver mode. It would be great, if gpstest would disable/stop the background measurements at that time to *save energy*.

**Describe alternatives you've considered**
* Hmm. Trying to remember to disable it always if I don't need it any longer running in the background? I tried! I failed at times.
* I might take another look, why my OS (LineageOS) does not stop the app when energy saver starts. I probably put gpstest on the exception list. I should check…

**Additional context**
Version: 3.10.2
OS: LineageOS 19.1

barbeau, 2024-01-23:
Not stale

gdt, 2024-04-12:
This makes sense, and I vote for "disable background service in energy saver mode" with no config.

ChristianTacke, 2024-04-12:
> This makes sense, and I vote for "disable background service in energy saver mode" with no config.

I don't really care, whether there is a config option or not. I added the "(optionaly)" to the title for the case that somebody thinks that this should be configurable.

In reality the energy saver mode setting itself is probably the "config option"? If you want to not save energy… well… disable the energy saver mode.

ChristianTacke, 2025-05-18:
Well, I think, this is still a valid feature request?
And I think, nobody made a claim that it's not valid?

Please un-stale?