barbeau/gpstest #652: Add support for B2b signal

Narugakuruga, 2023-08-15:
Mediatek Demisey 9200 add BDS-3 B2b and Galileo E5b signal supports.
B2b is broadcasted by all BDS-3 MEO/IGSO satellites (C19-C46) and PPP-B2b is broadcasted by BDS-3 GEO satellites (C59-C62).
Would be cool to see the new signal in action!

Narugakuruga, 2023-08-18:
Would be nice to test it out ☺️. 
I will check with a phone with D9200+ chipset someday

Narugakuruga, 2023-08-18:
I'm very curious to find out these Beidou-3 GEO satellites use B1l frequency for navigation service, B1C and B2a frequency for SBAS service. Their B2b is for high accuracy PPP measurement. 

While all these bands of other Beidou-3 sats is only for common navigation. 

So every GEO satellites serve 3 functions. That's interesting.

But right now the SBAS service is in test stage. Not for public use I think....

Narugakuruga, 2023-09-22:
[image]
Screenshot with Redmi K60 Ultra (D9200+ Soc)

Narugakuruga, 2023-10-05:
My test shows B2b is on 1207.14 MHz band. (Constellation type 5 for BDS right? )
Which is the old "B2/B2l" that gpstest already labeled. 
It's weird it doesn't shows up as B2 yet. 
[image]

Narugakuruga, 2023-10-05:
According to BDS official website.
B2b is expected to fully replace B2 as they use same 1207.14Mhz (maybe even un-interoperable too?)

barbeau, 2023-10-09:
>My test shows B2b is on 1207.14 MHz band. (Constellation type 5 for BDS right? )

Nice! What device is the screenshot from?

Narugakuruga, 2023-10-10:
Redmi K60 Ultra in China, aka Xiaomi 13T Pro :)
Maybe it's because android api couldn't report B2b signal? I guess.
So inside the GNSS chip it's using b2b to calculate location and PPP correction, but can't be read by android api?

Narugakuruga, 2023-10-10:
In GNSSLoger, BDS's 1207.14 MHz only shows up in MeasurementProviderMeasurement. 
But it would not display in Status SATELLITE_STATUS

Narugakuruga, 2023-10-24:
Ahh I see. Gnss logger can shows B2b normally.

Apparently mediatek 9200 chip only received B2b exclusively from 3 satellite. So it's easy to miss it.

Last time when I doing the measurement it was indoor, so I didn't receive the signal. 

So maybe Gpstest can show that too. We just have to rename B2 to B2b

[image]

barbeau, 2023-10-27:
Closed by https://github.com/barbeau/gpstest/pull/657

closed 2023-10-27 by barbeau