barbeau/gpstest #655: Vibrate / beep on location lock

JonnyTech, 2023-09-17:
**Is your feature request related to a problem? Please describe.**
My phone takes a long time to get a location lock - having a beep/vibrate to notify me would be very useful.

**Describe the solution you'd like**
Ability to set beep/vibrate upon lock - and possibly at regular intervals thereafter.

**Describe alternatives you've considered**
Checking the phone regularly, but sometimes can take several minutes.

**Additional context**
My eyesight is declining, having device feedback would remove the strain of focusing on the screen regularly.

Thanks for considering, it is not important, just nice to have :)

barbeau, 2023-09-24:
I agree, this would be useful in certain circumstances.

barbeau, 2024-04-14:
Not stale