barbeau/gpstest #656: More Compact Satellite Frequency status

Narugakuruga, 2023-09-22:
### Reason behind
As every GNSS constellation is using more frequency than ever before, the Satellite status page is kinda of bloated. 
For every 1 satellite, there can be up to 4 frequency to show up
For example Beidou B1/B1C/B2a/B2b
Galileo E1/E5a/E5b/E6
GPS L1/L2/L5
And there are already phones that support all the signal above (except E6).
[image]

### My suggestion
Idea 1
The easy way to reduce the visual clearness is, only show one ID for every physical satellite? It works when every satellite has the same frequency, and the interval between is just neat. But when SF and DF sats line up together it look like trash....

[image]

Idea 2
maybe we can compress the whole satellite, make the UI something like this?
but the CNR has no space to show. What if we use color to represent CNR,
for good signal above 35db, it will be pure green. For 30 to 34.9db it will be Brown green, and 20 to 29.9db yellow. And red below 20, and so forth.
But will require some coding.... I wish I could contribute if I can do coding.
 
Forgive my Photoshop technique😀 it's very rough idea. 
[image]

barbeau, 2023-09-25:
Yes, this is a tough one. The Filter and Sort features help with this, but the list does get long.

I've been working on a Dashboard feature I hope to release in the near future that is focused on more summary-level info that may help fill this gap:
https://github.com/barbeau/gpstest/pull/570

barbeau, 2024-04-14:
Closing this in favor of https://github.com/barbeau/gpstest/pull/570

closed 2024-04-14 by barbeau