barbeau/gpstest #658: [Early Concept] Detect code type (signal component) and additional navigation message

Narugakuruga, 2023-10-25:
GNSSMeasurement API provides code types. which reprecent different signal component.
https://developer.android.com/reference/android/location/GnssMeasurement#getCodeType()

For modern GNSS signal like L5, E5a, B2a, two component is provided for civilian use: 
### **I component, aka Data component.** 
The advantage of it sually contains addition navigation message different from L1 E1 B1 band. For example B2a use CNAV-2 message which only take 8s to receive a frame. While B1I and B1C take 30s.
B2b is one step further, its CNAV-3 message take 1s to receive. Similar things in Galileo and GPS.
However tracking this component means lower sensitivity and less accurate pseudo range.
### **Q component, aka Pilot component.**
This component exclusively contain pseudo range code. No navigation message is transmitted. It has better tracking performance and accuracy theoretically.

Therefore, also theoretically, GNSS chip should track I component only when it can utilize additional nav message.
Otherwise, track Q component for accuracy, and only decode L1 traditional nav message.

in GnssMeasurement getCodeType()
For L5-I, E5a-I, E5b-I, B2a-I, B2b-I, should output "I" or "D"
For L5-Q, E5a-Q, E5b-Q, B2a-Q, B2b-Q, should output "Q" or "P"

In my test, Qualcomm 870 appears to output Q for all L5 bands. And mediatek output I. Which means mediatek can decode addional CNAV-3 message. And it is in line with their advertisement "PPP-B2b"(Use addional data for accuracy correction)

For base L1/E1/B1I/L5 for IRNSS, this detection seens unnecessary. Because it's the mandatory requirement to decode traditional nav message, so tracking Data component is without choice.

Also GnssNavigationMessage provide a direct way to tell if there's additional nav message.
https://developer.android.com/reference/android/location/GnssNavigationMessage#TYPE_BDS_CNAV1

For GPS 
base:
TYPE_GPS_L1CA                                                                LNAV
addition:
TYPE_GPS_L2CNAV     L2C                                                L2-CNAV
TYPE_GPS_L5CNAV     L5                                                  L5-CNAV
TYPE_GPS_CNAV2       L1C                                                CNAV-2

For BDS
base
TYPE_BDS_D1             B1I (MEO/IGSO sat)                        D1
addition:
TYPE_BDS_D2             B1I (GEO sat)                                  D2
TYPE_BDS_CNAV1      B1C                                                 CNAV-1
TYPE_BDS_CNAV2      B2a                                                 CNAV-2
~~TYPE_BDS_CNAV3      B2b~~                                       CNAV-3       (google doesn't add it yet)

for Galileo
base:
TYPE_GAL_I                E1/E5b                                             I/NAV
addition:
TYPE_GAL_F               E5a                                                   F/NAV

For GLONASS 
only:
TYPE_GLO_L1CA        L1                                                      ??

For INRSS
only:
TYPE_IRN_L5CA         L5                                                      ??

For QZSS
only:
TYPE_QZS_L1CA        L1                                                       LNAV??

Narugakuruga, 2023-10-25:
Additional information: getCodeType() is vendor specific implemented. So obviously there are mistake. For example 
They output E1, L5(IRNSS) code type C means no data is transmitted. But obviously there's ephemeris data.

And my test shows SD870 receive ephemeris somewhat faster when use dual band. But it's ambiguous to measure.

I'm not sure if it's trustworthy to use this API

And I don't have devices to test GnssNavigationMessage

Narugakuruga, 2023-10-30:
@barbeau is it possible to lable the navigation message which device can decode?
As I discovered, most device never use BDS D2 message rather than D1.
So for BDS PNR 1-5, 59-61 satellite, they can never receive ephemeric data as they transmit different message. 
Also for QZSS PNR 196 satellite, there's no ephemeric or almanac at all in old device.

In newer device like Xiaomi 14 (SD 8 Gen 3), QZSS 196's almanac can be decoded finally. 
And for Mediatek 9200. BDS 59-61 message can be decoded to provide real world benefits (high accuracy clock and orbit correction)

barbeau, 2023-10-30:
>is it possible to lable the navigation message which device can decode?

Yes, I have this ticketed in https://github.com/barbeau/gpstest/issues/522.

Narugakuruga, 2023-10-31:
> > is it possible to lable the navigation message which device can decode?
> 
> Yes, I have this ticketed in #522.

Glad to know ! I may go to retail store for some testing and log someday