barbeau/gpstest #664: OSNMA support ?

Harvester57, 2024-01-19:
**Is your feature request related to a problem? Please describe.**
Implement support to compute OSNMA messages and check the integrity of the Galileo received signal

**Additional context**
More infos are available in the OSNMA ICD: https://[www.gsc-europa.eu](https://www.gsc-europa.eu/sites/default/files/sites/all/files/Galileo_OSNMA_SIS_ICD_v1.0.pdf)/sites/default/files/sites/all/files/Galileo_OSNMA_SIS_ICD_v1.0.pdf

gdt, 2024-04-12:
gpstest isn't really a "process raw data" client and I think should not cross into that.   Can you point to android APIs where there is anything about OSNMA?  To me, this request doesn't fit the scope of gpstest and should be closed, but maybe I'm missing something.

barbeau, 2024-05-19:
I'm open to adding this as part of spoofing detection. Data will come from the navigation message - https://developer.android.com/reference/android/location/GnssNavigationMessage

isarrider, 2024-06-21:
this is very interesting, especially for folks who suffer to live / be in areas where GNSS gets jammed / Spoofed...

buildvoc10, 2025-02-23:
If anyone is still interested in validating Open Source Navigation Message Authentication, just updating pic2bim https://github.com/buildvoc/PIC2BIMAndroid which already have this capabilities, (could this be added to GPSTest?) 

[image]

isarrider, 2025-03-25:
Some folks have been already implementing this - I found this description helpful...
https://github.com/daniestevez/galileo-osnma

muetzekoeln, 2025-07-23:
Tomorrow (july 24th 2025) OSNMA status will be set from "test" to "operational".

https://www.gsc-europa.eu/sites/default/files/sites/all/files/Galileo-service-notice-09-v1.2.pdf

gdt, 2025-09-02:
Jamming of government planes does not make OSNMA in GPSTest more relevant.    What's needed for GPSTest is a PR....

isarrider, 2026-10-29:
https://github.com/barbeau/gpstest/pull/864