barbeau/gpstest #668: Add cancel button on accuracy page

jidanni, 2024-02-05:
Here there is a save button, but not a cancel button.

Sure you could say well just hit your cell phone's back button. But that takes us out of the app.

Or you can say well just use close the app. But then we're just facing the same page the next time we open the app.

Sure you can say well just go into the menu and pick something else.

Well yes, but that's not all intuitive.
[image]

barbeau, 2024-02-16:
Thanks for the feedback! I think a more canonical Android way to do this would be to put the "X" at the end of the text entry box after you start typing to clear the box, so I'll leave this ticket open for that feature.

barbeau, 2024-02-16:
Gotcha - unfortunately that type of behavior doesn't match up with the standard Android navigation. You could say the same about any of the other tabs. So I won't be able to make that change.

jidanni, 2024-02-17:
Looking at random apps I find a Skip button,
[image]

You could call yours a No Thanks button.

jidanni, 2025-03-26:
Isn't it still valid?