barbeau/gpstest #669: Offer Openstreetmap etc

jidanni, 2024-02-05:
Here we see several choices, but they're all actually just a proprietary Google map.

Soon there's be some other choices, like Openstreetmap?

[image]

goldfndr, 2024-02-05:
#394

barbeau, 2024-02-16:
Thanks for the report! Yes, a discussion surrounding this is in https://github.com/barbeau/gpstest/issues/394. Basically, there is a separate OSS version that includes OSM maps. I'll keep this ticket open for adding OSM tiles as a map type to the google build flavor in case someone wants to take this on.

andreasscherbaum, 2025-03-23:
> I'll keep this ticket open