barbeau/gpstest #676: Suggestions on the display of satellite quantity and speed function.

xxbhpan, 2024-02-17:
These are some suggestions regarding my personal thoughts.

Firstly.
This app provides a satellite interface to display the relative position of satellites on a large disk.
I hope to add relevant statistical functions, such as how many GPS satellites there are, how many GLONASS satellites there are, how many BeiDou satellites there are, and so on. And it is best to display relevant information about the current usage/all numbers.

Secondly.
Some people want to use GPS tools to measure speed, and this app also provides related simplified functions.
This request may be somewhat offensive. I hope to add a speed interface that can display the current speed, direction (compass), distance traveled, top speed, and average speed in a simple way (UI), and can be freely paused and restarted. Provide relevant units such as km/h, m/s, MPH, KNOT, etc.

Thank you!

barbeau, 2024-02-19:
>I hope to add relevant statistical functions, such as how many GPS satellites there are, how many GLONASS satellites there are, how many BeiDou satellites there are, and so on. And it is best to display relevant information about the current usage/all numbers.

This is coming in the Dashboard tab I've outlined in https://github.com/barbeau/gpstest/issues/313 and partially implemented in https://github.com/barbeau/gpstest/pull/570.

>I hope to add a speed interface that can display the current speed, direction (compass), distance traveled, top speed, and average speed in a simple way (UI), and can be freely paused and restarted. Provide relevant units such as km/h, m/s, MPH, KNOT, etc.

I'm going to change the issue to focus on this simplified speed display. I think this could conceptually fit into the Dashboard too, but needs UI design, etc.

@xxbhpan Could you clarify if you plan to implement these features, or if you're just requesting that someone else work on them?

xxbhpan, 2024-02-19:
Thank you for your reply. I am very pleased that my first suggestion has been adopted and is about to be implemented.

Regarding the second question, due to the speed display on the current "Status" panel not being able to switch relevant units, I have even seen combinations written in some programs, such as   1.0m/s & 3.6km/h .

As you said, displaying complex information like what I proposed requires a new UI, but whether this will lead to the App being too complex will depend on you as a developer. However, simplified real-world speed information and the conversion of relevant speed units seem to be sufficient at present.

To be frank, if we can really implement the relevant functions on #570 , it would exceed the majority of apps developed around GPS functionality, even those that require payment. Thank you very much for your efforts.

barbeau, 2024-02-20:
>due to the speed display on the current "Status" panel not being able to switch relevant units, I have even seen combinations written in some programs, such as 1.0m/s & 3.6km/h .

Just to clarify, you can currently change the speed units in the settings:

[image]