barbeau/gpstest #678: [Enhancement] Apply network-delivered SBAS corrections

davecollett, 2024-03-19:
Related to the #617 request, but for SBAS messages.
At least EGNOS and SouthPAN provide network delivered versions of their SBAS messages via the internet (referred to as EDAS and SDAS respectively). These could be applied to the raw messages from the phone to potentially improve positioning accuracy. 
https://egnos.gsc-europa.eu/documents/egnos-data-access-service-sdd
https://www.linz.govt.nz/products-services/geodetic/southpan/accessing-southpan-services/service-definition

gdt, 2024-04-12:
It seems that processing raw is very much out of scope.   So this is about injecting SBAS data, and that requires an API.   Can you link to the Android API definition, if it exists?  If it doesn't exist, then this request isn't possible and should be closed.

gdt, 2024-05-19:
No API link has been posted, so I suggest closing this.

barbeau, 2024-05-19:
Looks like data is available over NTRIP, with details on the endpoints available after registration (see docs in first post).

barbeau, 2024-05-19:
True, this is predicated on GPSTest computing it's own positions from raw measurements, as mentioned in https://github.com/barbeau/gpstest/issues/617#issuecomment-1593657827, which it doesn't currently do.

GnssLogger computes WLS positions from raw measurements, so it's possible it could be in scope. Given the number of questions that arise around computing positions from raw measurements I'm open to it.

But it does require a significant amount of work to get there.

gdt, 2024-05-19:
That would be moving away from an articulated mission of "display the info from the system API so that you can understand what it's doing" and into "be some kind of general purpose GNSS toolkit including computing position solutions from raw".    I don't think any weight should be given to people asking for this; they seem not to understand the issues of API boundaries.

I do think an app that wraps RTKLIB etc. and computes not only pseudorange solutions but also RTK would be great.  I just don't see that it has an logical relation to what GPSTest is doing.   I don't see how blurring that would benefit either app.  It would certainly be bloat and a source of instability for GPSTest.

So I think you should say no firmly, and someday write that other app.