barbeau/gpstest #689: Satellites not showing up on Wahoo device

utack, 2024-07-31:
**Describe the bug**
On my Wahoo Android device the satellite list is empty, despite an existing fix

**To Reproduce**
Steps to reproduce the behavior:

1. Load GPSTest 3.10.3 from F-Droid onto a Wahoo Roam v2 device via adb
2. Start the App via adb monkey -p com.android.gpstest.osmdroid 1
3. Give permissions to the app (pm grant com.android.gpstest.osmdroid android.permission.ACCESS_FINE_LOCATION)

**Expected behavior**
The list of satellites shows up

**Observed behavior**
The list stays empty despite an existing GPS fix

**App, Device and Android version:** 

see above

**Screenshots:** 

[image]

barbeau, 2024-07-31:
Interesting! That device doesn't seem like an official Android release, so it probably just doesn't support the GnssStatus API that powers that UI.

Unfortunately there isn't anything I can do about this, you would have to ask the device manufacturer to add support for that API.

closed 2024-07-31 by barbeau

utack, 2024-07-31:
Unfortunate, thank you for checking!  
Is there any "adb" way to determine which constellations it supports or has a lock on?

utack, 2024-08-01:
Nope, there is not :(

```
NMEA,$GPGGA,081201.00,4805.0526,N,01133.0367,E,1,05,1.9,583.2,M,46.9,M,,*66,1722499920
NMEA,$GNGLL,4805.0526,N,01133.0367,E,081201.00,A,A*77,1722499920
NMEA,$GNGSA,A,3,67,73,81,,,,,,,,,,4.7,1.9,4.3,2*32,1722499920
NMEA,$GNGSA,A,3,27,32,,,,,,,,,,,4.7,1.9,4.3,1*39,1722499920
NMEA,$GNRMC,081201.00,A,4805.0526,N,01133.0367,E,1.7,297.0,010824,,,A,V*3F,1722499920
NMEA,$GNVTG,297.0,T,,M,1.7,N,3.1,K,A*1B,1722499920
NMEA,$GNZDA,081201.00,01,08,2024,,*7F,1722499920
NMEA,$PSEPU,19.4,9.4,7.0,6.3,17.0,17.57,8.84,90.5,6.34,6.16,15.18*45,1722499920
NMEA,$PSVLF,4.88,3.12,297.0,-2.78,1.41,-3.76*6C,1722499920
```