barbeau/gpstest #690: L1C is being labeled as L1 for GPS

barbeau, 2024-08-02:
**Describe the bug**

Reported in https://groups.google.com/g/gpstest_android/c/jAN2OL_KDX0/m/KxBsupWFBAAJ?utm_medium=email&utm_source=footer

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Run GPSTest on Xiaomi 14T Pro (Redmi K70 Ultra) with Mediatek 9300 

**Expected behavior**
L1C should be correctly labeled

**Observed behavior**
L1C is being labeled as L1

**App, Device and Android version:** 

Xiaomi 14T Pro (Redmi K70 Ultra) with Mediatek 9300

Narugakuruga, 2024-08-04:
So here is a sample of GPS L1C. These 3 piece of logs are from 2 GPS block III and 1 QZSS.
Apparently Mediatek is faithful to the Android API. L1C was correctly lable as "L". Which means right now the only two signals broadcast by one system sharing same frequency, can be determined now. 
L1C/A is "C", L1C is "L".
By the way I want to mention that Mediatek also use C for Galileo E1 (C), and the P for BDS B1C (Pilot). At the end of the day, it seems that everything was within the API documentation. While older device like Qualcomm chipset usually spam code "C" for every L1C/B1C band.
Which means that those manufacturer really started to care about the GNSS measurement proficiency.
[image]
[image]
[image]

Narugakuruga, 2024-08-04:
There're two more signals that share the same l1 band. QZSS L1C/B and L1S. Code type E and Z
We knew very little about l1cb, it is announced that the constellation will consist of l1ca and l1cb satellites. And l1cb is a PNT signal too. But whether the hardware or API will detect a code type difference is uncertain).
L1S is for high accuracy ppp position service. So it might not be that important

Narugakuruga, 2024-08-04:
Update: I think it's very unwise to include future signal when we cannot test and see results. 
And I think we could code a whitelist for every possible L1C code type, every L1CA code type for GPS and QZSS. The detailed list could be obtain from Android's document page.
The philosophy is if we encounter code type we didn't know and it's 1575.4 mhz, we could label it as "L1(?)" or put the name "1575.4 MHz"

Narugakuruga, 2025-05-12:
For GPS L1
| Signal  | component  | possible code| civilian purpose |
| ----  | ----  | ----  | ---- |
| L1C/A | C/A | C | ✅ |
| L1C | Data+Plot | S, L, X| ✅ |
| L1P(Y) | Data | P, Y | ❌ |
| L1M | | M | ❌ |

GPS L2
| Signal  | component  | possible code| civilian purpose |
| ----  | ----  | ----  | ---- |
| L2C/A | C/A | C | ✅ |
| L2C | Data+Plot | S, L, X| ✅ |
| L2P(Y) | Data | P, Y | ❌ |
| L2M | | M | ❌ |

GPS L5
| Signal  | component  | possible code| civilian purpose |
| ----  | ----  | ----  | ---- |
| L5 | I+Q | I, Q, X | ✅ |

Narugakuruga, 2025-05-12:
For QZSS L1
| Signal  | component  | possible code| civilian purpose |
| ----  | ----  | ----  | ---- |
| L1C/A | C/A | C | ✅ |
| L1C/B | C/B | E | ✅ |
| L1C | Data+Plot | S, L, X| ✅ |
| L1S | Data | Z | ✅ |
| L1Sb | Data | B | ✅ |

QZSS L5
| Signal  | component  | possible code| civilian purpose |
| ----  | ----  | ----  | ---- |
| L5 | I+Q | I, Q, X | ✅ |
| L5S | I+Q | D, P, Z | ❌ |

QZSS L6
| Signal  | component  | possible code| civilian purpose |
| ----  | ----  | ----  | ---- |
| L6D | Data+Plot | S, L, X | ✅ |
| L6E | E+Plot | E, Z | ✅ |

Narugakuruga, 2025-05-12:
For BDS B1
| Signal  | component  | possible code| civilian purpose | Note |
| ----  | ----  | ----  | ---- | ---- |
| B1I | I | I | ✅ | |
| B1C | Data+Plot | D, P, X| ✅ | Qualcomm device report C instead |
| B1Q | Q | Q | ❌ | |
| B1A | Data+Plot | S, L, Z | ❌ ||

BDS B2
| Signal  | component  | possible code| civilian purpose | Note |
| ----  | ----  | ----  | ---- | ---- |
| B2a | I+Q | D, P, X| ✅ | Qualcomm and Mediatek report I or Q |
| B2b | I+Q | D, P, Z | ✅ | Qualcomm and Mediatek report I or Q |
| B2 (ACE-BOC) | I+Q | D, P, X | ✅ | |
| B2I | I | I | ✅ and deprecated | |
| B2Q | Q | Q | ❌ and deprecated | |

BDS B3
| Signal  | component  | possible code| civilian purpose |
| ----  | ----  | ----  | ---- |
| B3I | I | I | ✅ |
| B3Q | Q | Q | ❌ |
| B3A | Data+Plot | D, P, Z | ❌ |