barbeau/gpstest #697: keypad "._" is broken on Accuracy page

JohnDenker, 2024-12-01:
When I press the ".-" key on the pop-up keypad, for entering lat/lon/elevation on the Accuracy page, no matter what I do, I get a "." not "-".

The workaround is to enter the number somewhere else and cut-and-paste, but that's a pain.

Suggestion: There seem to be unused keys on that keypad, so it might make sense to implement separate "." and "-" keys.

(Related suggestions will be filed as separate issues.)

====================================

App version: v3.10.3 (24097-google)
Model: SM-G955U
Android version: 9 / 28
Location: gps, hor acc 9.648001m, 1 second(s) ago
GNSS HW year: 2016
Raw measurements: NOT_SUPPORTED
Navigation messages: NOT_SUPPORTED
NMEA: SUPPORTED
Delete assist: NOT_SUPPORTED
Got fix: true
Dual frequency: NOT_SUPPORTED
Supported GNSS: GPS, GLONASS
Accumulated delta range: NOT_SUPPORTED
Automatic gain control: NOT_SUPPORTED
GNSS antenna info: NOT_SUPPORTED

    Google Play Services App: 24.44.33 (100400-694629214)
Google Play Services Library: 12451000

barbeau, 2024-12-02:
Hey @JohnDenker, could you share a screenshot of what you're seeing? The numeric keyboard is provided by the Android system so I don't have much control over that. Did you try long pressing on the ".-" key?

JohnDenker, 2024-12-02:
OK, new information:
- Just now I discovered that a **quick double-tap** on the ".-" key produces the "-" sign. This is unexpected on an Android platform. Normally a double-tap simply enters the primary meaning of the key, twice. Double-tap has a special meaning on many Apple platforms, but Android not so much.
- I observe that a long press enters the primary meaning of the key, i.e. ".". This is unexpected. Normally on an Android platform, a long press gives access to the secondary meaning(s) of the key.
- I observe that a single isolated tap enters the primary meaning of the key, as expected.
- A quick triple-tap is the same as a single tap, which is weird.

The screenshot shows the effect of four single taps, four long presses, four double-taps, and four triple-taps.

Suggestion; If you want to retain the existing double-tap behavior, that's fine ... but long press really should produce the expected secondary meaning.

[image]

barbeau, 2024-12-03:
Thanks for the additional info. AFAIK I can't control the behavior of the keyboard, that's up to your keyboard developer/device OEM if you're using the built-in keyboard. So I'd suggest to reach out to your keyboard developer for additional explanation.

closed 2024-12-03 by barbeau