barbeau/gpstest #699: Editing the reference point should not force a new log file.

JohnDenker, 2024-12-01:
When I edit the coordinates of the reference point on the Accuracy page and hit save -- even if I didn't change anything -- the app starts a new log file. Maybe sometimes that might be desirable, but usually not.

Remark: Orthogonality is a fundamental good UI principle. I see no connection between the reference point and the NMEA sentences I am logging to the file.

Observation: Often I am logging stuff even when I am not using the Accuracy feature.

Suggestion: There needs to be an explicit button (or some such) for rotating the log file. It should *not* be on the Accuracy page. Maybe somewhere under Settings.

barbeau, 2024-12-02:
@JohnDenker Thanks for the report! I can't reproduce this on a Galaxy 23 Ultra. Would you be able to share screenshots of your settings and ideally a video screen capture of this bug in action so I can see the exact steps your taking?

The log shouldn't be rotating when you edit the accuracy settings.

JohnDenker, 2024-12-02:
Let's close this issue as erroneous.
I will submit a new issue with a corrected subject line as well as corrected content.

closed 2024-12-02 by JohnDenker