barbeau/gpstest #712: Plot screen - signal list stops graphs from being displayed

smartnorman, 2025-03-26:
**Describe the bug**
Using GPSTest v3.10.5 on a prototype Trimble Android 14 handheld device, the default Plot screen is useless because so many signal labels are displayed that there's no room for graphs.  See attached Portrait and Landscape screen captures.

**To Reproduce**
Steps to reproduce the behavior:
1. Run GPSTest in Landscape mode
2. Select Plots
3. See that no graphs are visible on the Plots tab
4. Use Auto-rotate to change from Landscape to Portrait
5. See that no graphs are visible on the Plots tab

**Expected behavior**
SNR / CNo graphs should be visible on the Plots tab.

**Observed behavior**
No graphs are displayed on Plots tab.

**App, Device and Android version:** 
GPSTest v3.10.5 on a prototype Trimble Android 14 handheld device,
Google Play Services 25.09.34

**Screenshots:** 

[image]

[image]

barbeau, 2025-03-26:
Hey @smartnorman, thanks for the report! Looks like this is GnssLogger (not GPSTest) - could you please open an issue with this info at https://issuetracker.google.com/issues/new?component=313183&template=0 instead?

closed 2025-03-26 by barbeau

smartnorman, 2025-03-26:
Well, that's truly dumb!  Sorry.