barbeau/gpstest #716: Offer to Contribute Reproducible Test Case for Previously Reported Bug

Software-Testing600, 2025-05-23:
Hello!

I saw that the app previously had a bug (issue #330). I created a test (based on that version of the app) to help ensuring that the issue does not happen again. In other words, the test could be used for regression testing.

Would it be helpful if I contributed this test to your codebase? I’d be happy to open a pull request for the latest version, integrating the test with your existing test infrastructure.

You can find the script for issue #330 below:
```

import org.junit.Before;
import org.junit.Test;
import org.junit.runner.RunWith;

import android.content.Context;
import android.content.Intent;
import android.content.pm.PackageManager;
import android.content.pm.ResolveInfo;

import static androidx.test.core.app.ApplicationProvider.getApplicationContext;
import static androidx.test.platform.app.InstrumentationRegistry.getInstrumentation;

import androidx.test.filters.SdkSuppress;
import androidx.test.ext.junit.runners.AndroidJUnit4;
import androidx.test.uiautomator.By;
import androidx.test.uiautomator.UiDevice;
import androidx.test.uiautomator.UiObject2;
import androidx.test.uiautomator.Until;

import java.util.concurrent.TimeUnit;

import static org.hamcrest.CoreMatchers.notNullValue;
import static org.junit.Assert.assertThat;
import static org.junit.Assert.assertTrue;

/**
 * Regression test for GPSTest (com.android.gpstest).
 * Verifies that Help text properly explains accuracy flags.
 */
@RunWith(AndroidJUnit4.class)
@SdkSuppress(minSdkVersion = 18)
public class GpsTestHelpTextValidationTest {

    private static final String BASIC_SAMPLE_PACKAGE = "com.android.gpstest";
    private static final String INSTALLER_PACKAGE = "com.android.packageinstaller";
    private static final int LAUNCH_TIMEOUT = 5000;

    private UiDevice mDevice;

    @Before
    public void startMainActivityFromHomeScreen() {
        mDevice = UiDevice.getInstance(getInstrumentation());
        mDevice.pressHome();

        final String launcherPackage = getLauncherPackageName();
        assertThat(launcherPackage, notNullValue());
        mDevice.wait(Until.hasObject(By.pkg(launcherPackage).depth(0)), LAUNCH_TIMEOUT);

        // Launch the target app
        Context context = getApplicationContext();
        Intent intent = context.getPackageManager()
                .getLaunchIntentForPackage(BASIC_SAMPLE_PACKAGE);
        intent.addFlags(Intent.FLAG_ACTIVITY_CLEAR_TASK);
        context.startActivity(intent);

        mDevice.wait(Until.hasObject(By.pkg(BASIC_SAMPLE_PACKAGE).depth(0)), LAUNCH_TIMEOUT);
    }

    @Test
    public void testHelpTextIncludesAccuracyFlagExplanation() {
        try {
            TimeUnit.MILLISECONDS.sleep(5000); // wait for potential dialog
        } catch (InterruptedException e) {
            e.printStackTrace();
        }

        // Handle permissions and dialogs
        UiObject2 allowButton = mDevice.wait(Until.findObject(By.res(INSTALLER_PACKAGE, "permission_allow_button")), 2000);
        if (allowButton != null) allowButton.click();

        UiObject2 closeDialogButton = mDevice.wait(Until.findObject(By.res("android", "button3")), 2000);
        if (closeDialogButton != null) closeDialogButton.click();

        try {
            TimeUnit.MILLISECONDS.sleep(2000); // wait after dialogs
        } catch (InterruptedException e) {
            e.printStackTrace();
        }

        // Tap coordinates on screen to potentially dismiss overlays
        mDevice.click(75, 133);
        try {
            TimeUnit.MILLISECONDS.sleep(2000);
        } catch (InterruptedException e) {
            e.printStackTrace();
        }
        mDevice.click(75, 1500);

        // Open About section
        UiObject2 aboutButton = mDevice.wait(Until.findObject(By.text("About")), 2000);
        aboutButton.click();

        try {
            TimeUnit.MILLISECONDS.sleep(2000);
        } catch (InterruptedException e) {
            e.printStackTrace();
        }
        // Check content includes expected "E stands for 'estimated'" explanation
        UiDevice uiDevice = UiDevice.getInstance(getInstrumentation());

        UiObject2 helpTextElement = uiDevice.findObject(By.res("com.android.gpstest:id/help_text"));
        assertTrue(helpTextElement.getText().contains("E stands for 'estimated'"));
    }

    /**
     * Dynamically finds the current launcher app's package name.
     */
    private String getLauncherPackageName() {
        Intent intent = new Intent(Intent.ACTION_MAIN);
        intent.addCategory(Intent.CATEGORY_HOME);
        PackageManager pm = getApplicationContext().getPackageManager();
        ResolveInfo resolveInfo = pm.resolveActivity(intent, PackageManager.MATCH_DEFAULT_ONLY);
        return resolveInfo.activityInfo.packageName;
    }
}

```

Please let me know if you're open to this or if there are any contribution guidelines I should follow.

Thanks for your work on this project!

barbeau, 2025-05-23:
No thanks. Other more useful cases would be welcome.

Software-Testing600, 2025-05-24:
Thank you for your reply. Could you tell me why this test is not useful and what other types of test would be useful? I totally understand if you cannot get to this and thank you in advance for your time.

barbeau, 2025-05-24:
UI text like this can easily change and fail the test.

I'd prefer additional tests that cover crashes or computing correct values.

barbeau, 2025-05-24:
Also, I don't think the test as written above would even work