barbeau/gpstest #718: GPS summary widget

balaclava-guy, 2025-06-22:
**Is your feature request related to a problem? Please describe.**
I use my android phone as a bike computer, not knowing GPS strength, especially when commuting in populated cities is disadvantageous.

**Describe the solution you'd like**
Make available a widget (that can be resized to very small) which summarises (in whatever way appropriate) the current signal strength/satellite lock/GNSS status.

**Describe alternatives you've considered**
As far as I have researched, there are no other GPS apps that offer specifically signal strength as a widget (and certainly none that arent trying to collect/monetise data).

Thank you for an excellent app.

balaclava-guy, 2025-07-31:
@barbeau sorry, missed your response. Thanks for considering! Yes, this would be a home screen widget. Ideally we could start with something small and square (2x2? 4x4?). 

Some thoughts for an MVP below. I know very little about UX, and even less about GPS concepts/terminology, feel free to challenge.

#### Updating/Refreshing (Configurable)
- 'Every x seconds' (the default - to avoid consuming battery)
- 'Live' update (same as GPSTest feature that maintains lock outside of app + warning about battery usage)

#### UX (see mock-up)
- Show total number of connected satellites (top right of mock-up)
- Show connected constellations (if this is useful) [GP]S, [GA]lileo, [GL]ONASS, [BD]Beidou, [QZ]SS & [IR]SS
- Overall GPS 'strength' indicator = fn(numSatellitesConnected, numConstellationsAvailable,  ...) ... ?
- Follow [material design icon guidelines](https://developer.android.com/design/ui/mobile/guides/widgets/style): background with monochrome foreground (black/white for ease of design & legibility + dark/light mode compatibility

[image]

KaKi87, 2026-09-01:
Hi,

Any news on this please ?

Personally I'd like a widget for toggling GPSTest on and off, and seeing when the location is acquired, rather than monitoring over an extended period of time, my use case is having my location ready before opening Maps, which takes much longer to acquire on its own for some reason.

Thanks